xorbit/wesp32-demos #3: Van I use it in Arduino IDE

smartkodian, 2020-08-28:
Hi,

I worked with ESP32 and ENC80J26 shield with Temp Dallas and I used UIPEthernet library but because of power issue it's not useful and that's why I was looking for shield to work with PoE and I found yours and I think it's awesome.

My question is can I use my sketch in Arduino IDE with the same Library I used and compile it in wesp32 module without any issue or need another library?

And How can I order it to test it?

Thanks for your big effort.

xorbit, 2020-08-28:
Yes, there are several Arduino examples in this repo.  You need to select the wESP32 as board in the IDE, and `#include <ETH.h>`.  You need to use Espressif's ETH.h library to use the built-in Ethernet instead of the library you were using.

In the US you can order from Amazon: https://www.amazon.com/s?k=wesp32&ref=nb_sb_noss_2
Or order from Mouser: https://www.mouser.com/Search/Refine?Keyword=wesp32
Or from Tindie: https://www.tindie.com/search/?q=wesp32

smartkodian, 2020-09-10:
yes thank you. I tried the Olimex with ETH.h and work with me but don't have enough deocument and example of how to send data to my php server. Next week will try wESP32.

Is there any Document or example of how to send data to cloud server like php using http or https?

Thanks for great effort

xorbit, 2020-09-10:
Sorry, I create hardware but doing such a deep application dive is outside the scope of this repo.
There are plenty of guides online though, here's an example:

https://techtutorialsx.com/2017/11/18/esp32-arduino-https-get-request/

Just replace WiFi with ETH in the appropriate places.

closed 2020-09-10 by xorbit