xorbit/wesp32-demos #5: UDP example arduino sketch available? (needed for OSC sendbundle, Receive tests)

dappertutto, 2022-05-17:
Hi, I've looked through the examples for a UDP sketch, which generally I build off of for an OSC sketch like the one below. I've had success w/the Olimex PoE ISO and was hoping to test out this board. Any help with code would be very appreciated!

------

/*
Make an OSC bundle and send it over UDP

OSCBundles allow OSCMessages to be grouped together
to  preserve the order and completeness of related messages.
They also allow for timetags to be carried to represent the presentation time
of the messages.
 */
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <ETH.h>
#include <SPI.h>    
#include <OSCBundle.h>

EthernetUDP Udp;

//the Arduino's IP
IPAddress ip(192, 168, 1, 252);
//destination IP
IPAddress outIp(192, 168, 1, 125);
const unsigned int outPort = 9999;

 byte mac[] = {  
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; // you can find this written on the board of some Arduino Ethernets or shields
void setup() {
  ETH.begin();
  Udp.begin(8888);
}

void loop(){
  //declare the bundle
    OSCBundle bndl;

    //BOSCBundle's add' returns the OSCMessage so the message's 'add' can be composed together
    bndl.add("/analog/0").add((int32_t)analogRead(35));
    bndl.add("/analog/1").add((int32_t)analogRead(34));
    bndl.add("/digital/5").add((digitalRead(36)==HIGH)?"HIGH":"LOW");

    Udp.beginPacket(outIp, outPort);
        bndl.send(Udp); // send the bytes to the SLIP stream
    Udp.endPacket(); // mark the end of the OSC Packet
    bndl.empty(); // empty the bundle to free room for a new one

    bndl.add("/mouse/step").add((int32_t)analogRead(35)).add((int32_t)analogRead(34));
    bndl.add("/units").add("pixels");

    Udp.beginPacket(outIp, outPort);
        bndl.send(Udp); // send the bytes to the SLIP stream
    Udp.endPacket(); // mark the end of the OSC Packet
    bndl.empty(); // empty the bundle to free room for a new one

    delay(1000);
}

xorbit, 2022-05-17:
Would any of these examples be of use?

https://github.com/espressif/arduino-esp32/tree/master/libraries/AsyncUDP/examples

In general, the only thing you need to change is the physical layer connection from WiFi to Eth according to what I do in my examples, and the actual network traffic code should be the same whether it's connected over WiFi or Ethernet.

dappertutto, 2022-05-17:
Thanks for your reply. I hadn't encountered AsyncUDP before, just UDP. I'll give it a shot, then going over this in a couple of days with a friend that has a better understanding of the details than me. Appreciate the help!

dappertutto, 2022-05-20:
After a lot of testing, there might be some success, but I'll need to get a newer Wesp. The wifi functionality works fine, but for PoE, the Eth library is 115% of available memory, and worried about the old PHY connection. I see all of the improvvements in the last year, that's great! Thanks for the help.

closed 2022-05-20 by dappertutto