xorbit/wesp32-demos #6: Wesp board failure

bruingineer, 2022-06-28:
Hello,

I have been using the wesp32 board and think it is great!

A few days ago, the board stopped powering on. I do not see any visual indicators of a burned / bad component on the board. Is there any other way I can trouble shoot why the board won't power on?

Many thanks

xorbit, 2022-06-28:
Hi, can you tell me which revision the board is?
When you say it "stopped powering on", does that mean when powered from PoE or also from USB?  Can you still communicate with the ESP32 through a programmer / serial interface?

bruingineer, 2022-06-30:
I can turn it on with the usb programming interface. It seem as though powering through POE has failed

xorbit, 2022-07-02:
Odd.  First time I've heard of a PoE failure in the wild.  Which rev board?  Anything getting hot in the PoE section?