xorbit/wesp32-demos #7: some updated code and more detail on how to explore the WESP32

timbo100, 2023-04-04:
I just purchased several of the wESP32 board as I'm needing to use ESP32 with Ethernet.  I use esp idf v4.4 and I'm not familiar with initialization for ethernet or any other variant.  I was initially pleased to see that esp-idf vs arduino was being used but now I'm at a loss.  I don't do python, I use C++ The examples are over 5yrs old and I can't figure out how to use this board.  Please help.

This what would be helpful:
* assuming the python examples will work and employ espressif tools <=4.4 how do I explore it and run the examples--I don't know anything about the micropython environment or how that works.  I'm just used to the espidf tools, i.e. idf.py build etc.  And also use vscode.
* are there any up to date, esp-idf <=4.4, C++ examples I can explore?

Please help I can use a lot of these boards if I can development with them.   esp-idf >=5.0 has too many migration changes and is for another day.

tc@nuspatial.com   

Thanks

xorbit, 2023-04-04:
Hi Tim,
I saw your email as well but decided to answer here to avoid duplication and maybe benefit other users.

Short story is that there are no ESP-IDF examples in this repo because the ESP-IDF has Ethernet example code of its own that work fine with the wESP32 and I see no advantage in replicating any examples here that may get outdated and depend on the exact version of ESP-IDF used.  The only thing worse than no examples is examples that don't work IMO.  By relying on the Ethernet examples that are part of the ESP-IDF, they should always be in sync with the version of the IDF they are part of.

The [wESP32 website](https://wesp32.com/software/#esp-idf) shows how to configure the ESP-IDF examples correctly to work with the wESP32.  It also has a piece of code that shows what calls are needed to make it work in your own code.  This was last updated in November last year to work with IDF 4.

timbo100, 2023-04-05:
Thanks for being helpful.

closed 2023-04-08 by xorbit